ant0nkr/luxpower-ha-integration #16: Multiple Inverters?

Cortelhouse, 2025-07-25:
### Checklist

- [x] I have filled out the template to the best of my ability.
- [x] This only contains 1 feature request (if you have multiple feature requests, open one feature request for each feature request).
- [x] This issue is not a duplicate feature request of [previous feature requests](https://github.com/ludeeus/integration_blueprint/issues?q=is%3Aissue+label%3A%22Feature+Request%22+).

### Is your feature request related to a problem? Please describe.

I Do not see how this works with multiple Inverters

### Describe the solution you'd like

ability to have multiple inverters

### Describe alternatives you've considered

none

### Additional context

none

ant0nkr, 2025-07-26:
Hello @Cortelhouse ,
There is a setting Entity Prefix. Just set there a value and it will be added to Device and all entities will be starting with this prefix.

I attached a screenshot from my configuration, I have 3 inverters and all of them working at the same time just perfect:)

[image]

Thank you.

PluginMike, 2025-07-29:
Hi ant0nkr. Love the bridge BTW!

Just want to ask, is there anyway that i can pull the values together, like aggregating them. I have 2 in parralell and would like to use those metrics in the energy settings. 

Have a AWESOME day :D