ant0nkr/luxpower-ha-integration #20: Read Only Mode

PluginMike, 2025-07-29:
### Checklist

- [x] I have filled out the template to the best of my ability.
- [x] This only contains 1 feature request (if you have multiple feature requests, open one feature request for each feature request).
- [x] This issue is not a duplicate feature request of [previous feature requests](https://github.com/ludeeus/integration_blueprint/issues?q=is%3Aissue+label%3A%22Feature+Request%22+).

### Is your feature request related to a problem? Please describe.

I would like a surefire way to know that I can read the registers but NEVER write to them. 

### Describe the solution you'd like

A toggle to make the inverters run as read only. Or add a line into the configuration.yml

### Describe alternatives you've considered

Other LXP bridges, but they are not nearly as good as this one :)

### Additional context

N/A

ant0nkr, 2025-07-29:
Hello @PluginMike 

Nice to hear you like this integration :)

I'll try to add configuration parameter to make all editable entities read-only. From what I can see it is possible.

My plan is to add this into the next release which will happen next week.

Thank you.

PluginMike, 2025-07-29:
Thanks hey!

closed 2025-07-30 by ant0nkr