ant0nkr/luxpower-ha-integration #34: Lux SNA 6000 does not show sensors from the JK BMS via communication

vuong244, 2025-08-01:
### System Health details

[image]

### Checklist

- [x] I have enabled debug logging for my installation.
- [x] I have filled out the issue template to the best of my ability.
- [x] This issue only contains 1 issue (if you have multiple issues, open one issue for each issue).
- [x] This issue is not a duplicate issue of any [previous issues](https://github.com/ludeeus/integration_blueprint/issues?q=is%3Aissue+label%3A%22Bug%22+)..

### Describe the issue

I’m using the Lux SNA 6000 inverter and a JK BMS (model JK-PB2A16S20P) to communicate with the inverter. However, the battery information is not being displayed.

### Reproduction steps

1.
2.
3.
...

### Debug logs

```text
.
```

ant0nkr, 2025-08-01:
Hello @vuong244 ,
Does Inverter show information from JK BMS?
Most likely communication between Inverter and battery.

If Inverter shows information - then I need debug logs to analyze it.

Thank you.

ant0nkr, 2025-08-02:
Hi @vuong244 .

Thanks for the logs.
Just to confirm, it's never getting initialized, even after 5-10 minutes?

Thank you.

vuong244, 2025-08-02:
My wifi dongle is DG43904907

vuong244, 2025-08-02:
[image]
[image]

[home-assistant_lxp_modbus_2025-08-02T15-03-14.395Z.log](https://github.com/user-attachments/files/21559256/home-assistant_lxp_modbus_2025-08-02T15-03-14.395Z.log)

now it is not receiving data from the lux inverter

ant0nkr, 2025-08-02:
@vuong244 
Could you do a final test: register-block-size-test-3

It should be processing much better. 
As for some of the data I need to contact with LXP Dev support, hopefully they will help. Since there is an issue in responses. they looks valid, but they are mixed up.

Thanks a lot for your help!

ChristopheKoeck, 2025-08-02:
Hi @ant0nkr 

I have tried test-3 and i get the same result as @vuong244 

Let me know if you need any more logs or info about either the dongle or inverter.

Thanks for looking into this for us!

ChristopheKoeck, 2025-08-02:
@ant0nkr not sure if this matters, but looks like @passantino and me don't have the standard wifi dongle, we have the newer ewifi dongle. It's supposed to be an updated version of the normal wifi dongle.

vuong244, 2025-08-02:
Thank you for your enthusiastic support. but it seems like it still doesn't work

[home-assistant_lxp_modbus_2025-08-02T23-15-21.813Z.log](https://github.com/user-attachments/files/21563001/home-assistant_lxp_modbus_2025-08-02T23-15-21.813Z.log)

ant0nkr, 2025-08-06:
@passantino tcp dump would help, from/to eWifi dongle, when it's communicating with official app. I'm curious what is actually being sent/received. 

Luxpower support is not very helpful right now, but I'm trying to convince them to send me documentation for wifi/e-wifi dongles :)

It can be done via Wireshark or by router itself, but it depends on a network configuration. Then you would need to use official app and make different requests, changing some params or reading them. It will generate set of requests/responses and I can try to decode them and find out what is the difference between wifi and eWifi dongles.

Thank you.

vuong244, 2025-08-06:
> Unfortunately LXP support didn't share "secret" document how to work with eWifi dongles.
> 
> I'll continue to research on how to send packets correctly.
> 
> Thank you.

Hello @ant0nkr 
I already have an integration that retrieves the inverter and battery parameters. However, it has fewer configuration features compared to your integration, and the original developer no longer maintains or updates it. I wonder if you're interested in it to possibly enhance your own integration?

ant0nkr, 2025-08-06:
Hello @vuong244 
Yes, I'm interested in enhancing integration and especially extending support for more dongles.

vuong244, 2025-08-06:
> Hello [@vuong244](https://github.com/vuong244) Yes, I'm interested in enhancing integration and especially extending support for more dongles.

Hello @ant0nkr 
How should I send it to you? Please give me your email address. The person who wrote this integration doesn’t want me to upload it here.

ant0nkr, 2025-08-06:
@vuong244 you can send to: kramskoy.anton+github@gmail.com

Let me know when you send, I'll remove it

vuong244, 2025-08-06:
hello @ant0nkr 
Please find the attached log file

[home-assistant_lxp_modbus_2025-08-06T12-45-30.077Z.log](https://github.com/user-attachments/files/21618654/home-assistant_lxp_modbus_2025-08-06T12-45-30.077Z.log)

[image]

ant0nkr, 2025-08-06:
@vuong244 you have downloaded main branch and restarted Home Assistant?
I don't see 1 change in the requests, which confuses me a bit.

vuong244, 2025-08-06:
@ant0nkr i I downloaded and installed version 0.1.5

ant0nkr, 2025-08-06:
@vuong244 
Could you try 0.1.6-beta, please.

vuong244, 2025-08-06:
> [@vuong244](https://github.com/vuong244) Could you try 0.1.6-beta, please.

@ant0nkr 
That's great, it worked successfully. Thank you!

ant0nkr, 2025-08-06:
So.. Can you confirm issue is closed and there is no more missing data?

Could you send me logs in any case, please.

Thank you!

ant0nkr, 2025-08-06:
@vuong244 it is on HACS: https://github.com/ant0nkr/luxpower-ha-integration/releases/tag/bg-dongle-test-release

@ChristopheKoeck great to hear! We're close to get it solved :) 

I think, bg-dongle-test-release should close all sensors.

ChristopheKoeck, 2025-08-06:
@ant0nkr i've tried the bg dongle version, that one doesn't work for me.
Nothing gets filled in. The 0.1.6 did work

ChristopheKoeck, 2025-08-06:
> [@passantino](https://github.com/passantino) tcp dump would help, from/to eWifi dongle, when it's communicating with official app. I'm curious what is actually being sent/received.
>
> Luxpower support [...]

@ant0nkr when you have time, can you explain to me how to do this in simple steps. what software i need and which steps to take.
or if there is a source for this already can you point me in the right direction?

ChristopheKoeck, 2025-08-06:
I'm also missing PV2 Energy Today it seems.

[image]

ant0nkr, 2025-08-06:
@ChristopheKoeck is it from 0.1.6-beta? Could you send me the logs from HA. I'll review them.

Right now, there is no need in wireshark, I found a way to obtain them :)

Thank you!

vuong244, 2025-08-07:
hello @ant0nkr 
There are still a few parameters missing: Load Consumption Today, Load Consumption Total, Load Power, and Solar Production Today.
Thank you very much

vuong244, 2025-08-13:
Hello @ant0nkr 
In that case, could you add a quick-charge button to my integration, like the quick-charge button in your integration?
Thanks you

dstark0011, 2025-08-13:
Morning gents, 

I'm experiencing almost the same issue.  Would you like me to create a new issue, with the full details on my problem, or is it ok to discuss on this thread?

Currently running version 0.1.6b. Had the same issue with 0.1.5.

ant0nkr, 2025-08-15:
> Morning gents,
> 
> I had the same issue - resolved with the bg-dongle-test release!!
> 
> Thank you so much.

Hello @dstark0011 
Did bg-dongle-test resolved all your issues?
That's a good news. I would need to add then logic placed in bg-dongle-test.

vuong244, 2025-08-15:
> > Hello [@ant0nkr](https://github.com/ant0nkr) In that case, could you add a quick-charge button to my integration, like the quick-charge button in your integration? Thanks you
> 
> Hello [@vuong244](https://github.com/vuong244) You can try to install quick charge blueprint. How to install available in documentation.

hello @ant0nkr 
I can’t find the AC Charge Type Select in my integration to enable charging mode.
Thanks you

ant0nkr, 2025-08-17:
Hi @vuong244 
Could you show me all available data in HA for your inverter? I'd like to see what's not being shown.
And, if possible, send me HA logs, so that I could confirm the issue if there is.

ChristopheKoeck, 2025-08-19:
I've been using the 0.1.6 beta for the last 2 weeks and it's mostly working.
I'm really only missing PV2 Daily Total to get my Energy Dashboard working

ant0nkr, 2025-08-21:
@ChristopheKoeck 
I see it is pretty good.
I'm not sure what inverter you have, but I see PV1 and PV. PV3/PV4 may not be returned, just because they do not exists in your inverter :)

ChristopheKoeck, 2025-08-22:
@ant0nkr that's correct.
I only have PV1 and PV2. I just wish that PV Energy Today would be there like PV1 Energy Today. I need to total of those to calculate my total PV Energy for the day.
I have the LuxPower 12KW inverter.