ant0nkr/luxpower-ha-integration #52: Invalid responses from dongle

ivanfmartinez, 2025-09-30:
### System Health details

## System Information

version | core-2025.9.4
-- | --
installation_type | Home Assistant Container
dev | false
hassio | false
docker | true
container_arch | amd64
user | root
virtualenv | false
python_version | 3.13.7
os_name | Linux
os_version | 6.11.3-200.fc40.x86_64
arch | x86_64
timezone | America/Recife
config_dir | /config

<details><summary>Home Assistant Community Store</summary>

GitHub API | ok
-- | --
GitHub Content | ok
GitHub Web | ok
HACS Data | ok
GitHub API Calls Remaining | 5000
Installed Version | 2.0.5
Stage | running
Available Repositories | 2272
Downloaded Repositories | 36

</details>

<details><summary>Solcast Solar</summary>

can_reach_server | ok
-- | --

</details>

<details><summary>Home Assistant Cloud</summary>

logged_in | false
-- | --
can_reach_cert_server | ok
can_reach_cloud_auth | ok
can_reach_cloud | ok

</details>

<details><summary>Dashboards</summary>

dashboards | 8
-- | --
resources | 20
views | 62
mode | storage

</details>

<details><summary>Recorder</summary>

oldest_recorder_run | September 7, 2025 at 16:54
-- | --
current_recorder_run | September 29, 2025 at 21:49
estimated_db_size | 5146.73 MiB
database_engine | sqlite
database_version | 3.48.0

</details>

<details><summary>Sonoff</summary>

version | 3.9.3 (7c75c46)
-- | --
cloud_online | 1 / 12
local_online | 1 / 1

</details>

### Checklist

- [x] I have enabled debug logging for my installation.
- [x] I have filled out the issue template to the best of my ability.
- [x] This issue only contains 1 issue (if you have multiple issues, open one issue for each issue).
- [x] This issue is not a duplicate issue of any [previous issues](https://github.com/ludeeus/integration_blueprint/issues?q=is%3Aissue+label%3A%22Bug%22+)..

### Describe the issue

after initial tests where I saw some data I found that some entities are unknown and some that came from INPUT have wrong data .

Adding more debug I found that my inverter is giving broken responses and this is mixing the response values. https://github.com/ant0nkr/luxpower-ha-integration/pull/51

I have tried with 40 and 125 bytes. In both cases mostly on INPUT registers the responses came with less than the expected registers

### Reproduction steps

1. Activate and wait for polled data
2.
3.
...

### Debug logs

```text
To help understand I have added more checks and logs to the modbus_client
 Here some samples 

2025-09-29 21:50:23.156 DEBUG (MainThread) [custom_components.lxp_modbus.classes.modbus_client] Polling INPUT 0-39: Req[38]: a11a0100200001c2444734333930383135341200000434333934305430303934000028002ade, Resp[117/117]: a11a0500bf0001c244473433393038313534b1000003343339343054303039340000a0859a0900343339343054303039344541414203212201010019091d15320d01000100000000002c000000f5f3d007b400b400e10706093e17841745072e0afa0064008b04b90a32000200d101b90a02000200

2025-09-29 21:50:23.156 DEBUG (MainThread) [custom_components.lxp_modbus.classes.modbus_client] input_response 0/40 dict={}

2025-09-29 21:50:23.156 ERROR (MainThread) [custom_components.lxp_modbus.classes.modbus_client] Invalid INPUT registers response expected=40 received=0 response[117] a11a0500bf0001c244473433393038313534b1000003343339343054303039340000a0859a0900343339343054303039344541414203212201010019091d15320d01000100000000002c000000f5f3d007b400b400e10706093e17841745072e0afa0064008b04b90a32000200d101b90a02000200:

2025-09-29 21:52:35.767 DEBUG (MainThread) [custom_components.lxp_modbus.classes.modbus_client] Polling INPUT 0-124: Req[38]: a11a0100200001c244473433393038313534120000043433393430543030393400007d00158e, Resp[41/287]: a11a0500230001c244473433393038313534150000033433393430543030393407000445414142db62

2025-09-29 21:52:35.767 DEBUG (MainThread) [custom_components.lxp_modbus.classes.modbus_client] input_response 2/125 dict={7: 16709, 8: 16961}
2025-09-29 21:52:35.767 ERROR (MainThread) [custom_components.lxp_modbus.classes.modbus_client] Invalid INPUT registers response expected=125 received=2 response[41] a11a0500230001c244473433393038313534150000033433393430543030393407000445414142db62:

2025-09-29 21:52:35.976 DEBUG (MainThread) [custom_components.lxp_modbus.classes.modbus_client] Polling HOLD 0-124: Req[38]: a11a0100200001c244473433393038313534120000033433393430543030393400007d00528c, Resp[197/287]: a11a0500bf0001c244473433393038313534b1000003343339343054303039340000a0859a0900343339343054303039344541414203212201010019091d15341901000100000000002c000000f5f3d007b400b400e10706093e17841745072e0afa0064008b04b90a32000200d101b90a020002002e0a6c167418f401e8033a16a6180a000a003a16a6180a000a0000006b08e807c4084709000064006400e803e8036400640005006400090010000000000000000000640064000000173b000000000ba0

2025-09-29 21:52:35.976 DEBUG (MainThread) [custom_components.lxp_modbus.classes.modbus_client] hold_response 80/125 dict={0: 39557, 1: 9, 2: 13108, 3: 13369, 4: 21552, 5: 12336, 6: 13369, 7: 16709, 8: 16961, 9: 8451, 10: 290, 11: 1, 12: 2329, 13: 5405, 14: 6452, 15: 1, 16: 1, 17: 0, 18: 0, 19: 44, 20: 0, 21: 62453, 22: 2000, 23: 180, 24: 180, 25: 2017, 26: 2310, 27: 5950, 28: 6020, 29: 1861, 30: 2606, 31: 250, 32: 100, 33: 1163, 34: 2745, 35: 50, 36: 2, 37: 465, 38: 2745, 39: 2, 40: 2, 41: 2606, 42: 5740, 43: 6260, 44: 500, 45: 1000, 46: 5690, 47: 6310, 48: 10, 49: 10, 50: 5690, 51: 6310, 52: 10, 53: 10, 54: 0, 55: 2155, 56: 2024, 57: 2244, 58: 2375, 59: 0, 60: 100, 61: 100, 62: 1000, 63: 1000, 64: 100, 65: 100, 66: 5, 67: 100, 68: 9, 69: 16, 70: 0, 71: 0, 72: 0, 73: 0, 74: 100, 75: 100, 76: 0, 77: 15127, 78: 0, 79: 0}

2025-09-29 21:52:35.976 ERROR (MainThread) [custom_components.lxp_modbus.classes.modbus_client] Invalid HOLD registers response expected=125 received=80 response[197] a11a0500bf0001c244473433393038313534b1000003343339343054303039340000a0859a0900343339343054303039344541414203212201010019091d15341901000100000000002c000000f5f3d007b400b400e10706093e17841745072e0afa0064008b04b90a32000200d101b90a020002002e0a6c167418f401e8033a16a6180a000a003a16a6180a000a0000006b08e807c4084709000064006400e803e8036400640005006400090010000000000000000000640064000000173b000000000ba0:

2025-09-29 21:55:48.900 DEBUG (MainThread) [custom_components.lxp_modbus.classes.modbus_client] Polling INPUT 0-124: Req[38]: a11a0100200001c244473433393038313534120000043433393430543030393400007d00158e, Resp[197/287]: a11a0500bf0001c244473433393038313534b1000003343339343054303039340000a0859a0900343339343054303039344541414203212201010019091d15372601000100000000002c000000f5f3d007b400b400e10706093e17841745072e0afa0064008b04b90a32000200d101b90a020002002e0a6c167418f401e8033a16a6180a000a003a16a6180a000a0000006b08e807c4084709000064006400e803e8036400640005006400090010000000000000000000640064000000173b000000009c92

2025-09-29 21:55:48.900 DEBUG (MainThread) [custom_components.lxp_modbus.classes.modbus_client] input_response 80/125 dict={0: 39557, 1: 9, 2: 13108, 3: 13369, 4: 21552, 5: 12336, 6: 13369, 7: 16709, 8: 16961, 9: 8451, 10: 290, 11: 1, 12: 2329, 13: 5405, 14: 9783, 15: 1, 16: 1, 17: 0, 18: 0, 19: 44, 20: 0, 21: 62453, 22: 2000, 23: 180, 24: 180, 25: 2017, 26: 2310, 27: 5950, 28: 6020, 29: 1861, 30: 2606, 31: 250, 32: 100, 33: 1163, 34: 2745, 35: 50, 36: 2, 37: 465, 38: 2745, 39: 2, 40: 2, 41: 2606, 42: 5740, 43: 6260, 44: 500, 45: 1000, 46: 5690, 47: 6310, 48: 10, 49: 10, 50: 5690, 51: 6310, 52: 10, 53: 10, 54: 0, 55: 2155, 56: 2024, 57: 2244, 58: 2375, 59: 0, 60: 100, 61: 100, 62: 1000, 63: 1000, 64: 100, 65: 100, 66: 5, 67: 100, 68: 9, 69: 16, 70: 0, 71: 0, 72: 0, 73: 0, 74: 100, 75: 100, 76: 0, 77: 15127, 78: 0, 79: 0}

2025-09-29 21:55:48.900 ERROR (MainThread) [custom_components.lxp_modbus.classes.modbus_client] Invalid INPUT registers response expected=125 received=80 response[197] a11a0500bf0001c244473433393038313534b1000003343339343054303039340000a0859a0900343339343054303039344541414203212201010019091d15372601000100000000002c000000f5f3d007b400b400e10706093e17841745072e0afa0064008b04b90a32000200d101b90a020002002e0a6c167418f401e8033a16a6180a000a003a16a6180a000a0000006b08e807c4084709000064006400e803e8036400640005006400090010000000000000000000640064000000173b000000009c92:
```

ant0nkr, 2025-09-30:
Is it happening all the times or randomly?

ivanfmartinez, 2025-09-30:
> Is it happening all the times or randomly?

I don't saw in the logs it reading more than 80 INPUT registers of first block anytime, the integration requests 125 and got only 80 normally. On the second block (125-249) it gives the correct size response.

 If I change to use 40 registers the responses are more unstable and also create problem with holding. I have tried some changes without results. Aparently the response is valid but only with less registers than requested.

Maybe I will be able to do more carefull tests on weekend, I opened the issue to see if someone else have anything like this.

I forgot to write that the inverter is LXP-LB-BR (apparently it uses same firmware of US version). I don´t know if this is a inverter or dongle response truncation error.

ant0nkr, 2025-10-01:
@ivanfmartinez 
The same. V23.
However, it does not describe how to work with dongle, it describes how to work directly with Luxpower inverter using RJ45->USB converter. I'm working on getting one and trying to do a direct communcation instead of using dongle, but this is in progress.

Dongle requires some additional info passed in order to work and LXP Dev team doesn't want to share it :)

ivanfmartinez, 2025-10-03:
Apparently dongle has a bug (logs added to get more information)

40 registers are requested, the response expected to be 117 bytes, after reading and decoding that the packet indicated 80 more bytes than got, this indicated that probably the dongle responded with 80 registers instead of 40 because on next read only 80 bytes came without the A11A header.

Need to change the current logic to bypass this condition. 

Also the current code ignore all data if a single request failed, probably using the valid received data is better.

```
2025-10-03 08:01:03.046 DEBUG (MainThread) [custom_components.lxp_modbus.classes.modbus_client] Polling INPUT 0-39: Req[38]: a11a0100200001c2444734333930373236321200000434333934305430303039000028000786, Resp[117]: a11a0500bf0001c244473433393037323632b1000003343339343054303030390000a0859809003433393430543030303945414142032525010100190a0308003b01000100000000002c00000075f3d007b400b4002e08000a3e178417e006a009fa0064004c04240a32000200b801240a02000200
2025-10-03 08:01:03.046 DEBUG (MainThread) [custom_components.lxp_modbus.classes.modbus_client] ignoring INPUT packet for regs 0-39 : packet_error=True/Wrong packet length expected=197 received=117 register=-1 function=-1
2025-10-03 08:01:03.046 DEBUG (MainThread) [custom_components.lxp_modbus.classes.modbus_client] Polling INPUT 40-79: Req[38]: a11a0100200001c2444734333930373236321200000434333934305430303039280028000e26, Resp[80]: a0096c167418f401e8033a16a6180a000a003a16a6180a000a0000006c08e807c4084809000064006400e803e8036400640005006400000000000000000000000000640064000000173b000000008750
2025-10-03 08:01:03.046 DEBUG (MainThread) [custom_components.lxp_modbus.classes.modbus_client] ignoring INPUT packet for regs 40-79 : packet_error=True/Missing A11A header register=-1 function=-1

```

closed 2025-10-04 by ivanfmartinez